dakboard/Hardware-OS #7: Dakboard Registration Serial Number (0)

zerotorr, 2021-01-21:
I'm attempting to register Dakboard on a raspberry pi, but after logging onto my network, the dakboard asks me to add a "Displays and Devices" with a serial number of "0" 

How do I get a unique serial number?

MrDadaGuy, 2022-01-15:
Thanks for the solution, but have to agree with above - this should be documented much more clearly - very frustrating.

Kurtman66, 2022-11-24:
Same issue here.  The url appearing now (from @danking6 ) seems to show instructions for "Raspberry Pi - Download and Install the DAKboard OS", but I don't actually see anything re: obtaining a unique serial number here...  any further advice when your DAKB says "serial #0"?  Thanks all...