dakboard/Hardware-OS #11: Display webpage

ryan8382, 2021-01-26:
Please add more options to the website management. 

- Static IP address/DNS

- Reboot, Shutdown

- Show IP info, MAC Address

- Network prioritization

- Wifi signal strength

I'm sure there are other items that could be added.

cocarrig, 2021-02-17:
Wireless strength indication added in #1.
Reboot capability has defined use cases in #5 

Keeping this open and adding enhancement label.

cocarrig, 2022-01-12:
Converting this into a discussion, for the purpose of retaining the following requests:
Displaying the following in the configuration page of DAKboard OS:
- Ability to Set Static IP and DNS
- Show current IP and Mac address.

Network prioritization is a being made for hardline connection in the following case:
https://github.com/dakboard/Hardware-OS/issues/41