dakboard/Hardware-OS #24: Upgrade OS path

klamming, 2021-02-23:
**Is your feature request related to a problem? Please describe.**
I am adding a new device to my account and downloaded a new img file and noticed its a new version.  All my other screens are 2.16 and the new img is 2.20.  I would think an easy upgrade should be available from each device.  Maybe an upgrade script?

ScottAEvans, 2021-03-18:
Agreed with this, as I'm not seeing any built-in mechanism to keep the OS up to date.  I did try an apt-get upgrade on a test instance and it basically bricked my Pi.  Not sure if I did something wrong (I doubt it, since it's the same command I run on my other pis with no problems), but I didn't try it a second time to try and recreate the issue.  Not a big deal as I just reloaded the original image onto my microSD card, but there should be an easy way to keep at least the OS up to date, since dakboard is really just a browser sitting on top of the pi OS.

Narflar, 2021-12-03:
this would also be great so you don't have to remove the SD card and reflash it.  Keeping the Dakboard where it is and not having to screw with the SD Card would make things a lot easier.

UXgoesfirst, 2021-12-11:
I also tried the standard way via apt-get ... and bricked my pi as well.

cocarrig, 2022-01-06:
I will be converting this into a discussion, as this is a long term goal for the DAKboard OS project.