dakboard/Hardware-OS #30: Power off button

Dimasua, 2021-08-29:
I would like to have a button connected to the Raspberry Pi to be able to safely shutdown the Dakboard.

To do that just one line of the following code is required: _dtoverlay=gpio-shutdown_ in the file _/boot/config.txt_
A button is connected to the physical pin 5 (BCM 3) and the ground. It will switch off/on the Dakboard.
That is a pre-defined functionality of the Raspberry Pi OS.

cocarrig, 2022-09-21:
Included in 3.1 release code, written and maintained by the device configuration page automatically.  Release is pending UAT.

cocarrig, 2022-09-26:
Completed in DAKboard OS 3.10 now released:
https://github.com/dakboard/Hardware-OS/releases/tag/3.10
https://dakboard.freshdesk.com/support/solutions/articles/35000125880-download-and-install-the-dakboard-os-for-raspberry-pi

Dimasua, 2022-11-10:
Hi, great news! Is there any instruction which pins should be used for the shutdown of the Dakboard.

cocarrig, 2022-11-10:
Hi @Dimasua,

This was implemented exactly as requested, and power functions are performed from connecting a button on the GPIO 6 and 6 pin.  We have however identified that the power function is only functional for power on, and not power off so this case is being reopened.

Solution being considered:
https://github.com/Howchoo/pi-power-button

CommanderGL, 2023-10-14:
when do maintainers ever go back to old issues.

cocarrig, 2023-10-14:
@CommanderGL,

Fairly regularly, but less frequently in the Hardware-OS side of the github project, and whenever a new comment or idea is posted.

For this issue specifically, since DAKboard shop items arent using this functionality and instead rely on sleep schedules for most users this is somewhat low on the priority list for us at the moment, but we would still like to implement when extra development cycles are available.

If you need a solution sooner, you can gain privileged access to the OS once SSL is enabled using the following steps, then you can implement the button behavior in a manner that is supportive for your setup:
https://dakboard.freshdesk.com/support/solutions/articles/35000222207-dakboard-os-ssl-configuration-and-use