dakboard/Hardware-OS #44: DAKboard OS Settings Page Authentication

mjnichols1, 2022-02-17:
**Is your feature request related to a problem? Please describe.**
As a user, I would like the settings page for DAKboard OS configuration to be secured. Currently, you can access the settings page by using the IP in a browser only.

**Describe the solution you'd like**
- A login prompt (could be generic/default) but should be configurable if a default is applied

cocarrig, 2022-05-25:
https://www.cyberciti.biz/tips/lighttpd-setup-a-password-protected-directory-directories.html

cocarrig, 2022-09-20:
Path forward being discussed:
- Add collapsable 'Advanced' section at the bottom of the device configuration page, just before save.  Expanding reveals a toggle option to 'Password Protect Config Page', enabling reveals current set password field which can be updated.  Default user is admin (not configurable, shown gray) and password is 'dakb5142'.

Authentication toggle will comment/uncomment mod_auth in **/etc/lighttpd/lighttpd.conf**:

server.modules += (
        "mod_dirlisting",
        "mod_staticfile",
        **"mod_auth",**
)

Password updates will update the password via the following:
sudo htpasswd /etc/lighttpd/.htpasswd admin