dakboard/Hardware-OS #68: Raspberry OS Image 3.0 is missing CEC Utils

digitaldex, 2022-09-09:
The Scheduler for a display based on RPi image is not working. HDMI script in /home/pi complains about missing CEC Utils.

Workaround:
login via ssh an execute ‚sudo apt-get -y install cec-utils‘

cocarrig, 2022-09-20:
@digitaldex,

Thank you for reporting this, it looks like since the bullseye release development(2.32-3.0) originally predated the addition of CEC-utils from 2.29 this was not in the image.  This will be added back in our next release.  Fortunately this is only impacting script feedback, since vcgencmd is doing the work.  Thanks again!

digitaldex, 2022-09-20:
For me it was not only script feedback. The Scheduler didnt work without the utils. After manually installing cec-utils the TV went to sleep as expected.

cocarrig, 2022-09-20:
@digitaldex,

That is very interesting and unexpected, apologies.  Thanks for the heads up and we will get the package into our next release.

cocarrig, 2022-09-26:
Completed in DAKboard OS 3.10:
https://github.com/dakboard/Hardware-OS/releases/tag/3.10
https://dakboard.freshdesk.com/support/solutions/articles/35000125880-download-and-install-the-dakboard-os-for-raspberry-pi

closed 2022-09-26 by cocarrig