dakboard/Hardware-OS #83: serial number 0

maximillionfrday, 2023-02-14:
I am getting serial number 0.  can you assist?

cocarrig, 2023-02-14:
Hi @maximillionfrday,

This sounds like you are using the legacy DAKboard OS, which will need a serial number and ID added to the device manually that is generated on dakboard.com via the following instructions:
https://dakboard.freshdesk.com/support/solutions/articles/35000201491-legacy-dakboard-os-instructions

The legacy instructions should only be used for the Raspberry Pi Zero W 1, if using any other Raspberry Pi supported, please use the new DAKboard OS and the serial and ID is not needed:
https://dakboard.freshdesk.com/support/solutions/articles/35000125880-raspberry-pi-download-and-install-the-dakboard-os

If you are still having trouble, reach out to us here and we can review your specific setup with you:
https://dakboard.freshdesk.com/support/tickets/new

closed 2023-02-14 by cocarrig

maximillionfrday, 2023-02-14:
yeah I chose that one on purpose because I was using a Raspberry Pi 3 not a zero however I did want the allocation of more time I settings for that for the Raspberry Pi zero and the older OS

 originally I was using the most current OS,  however, I was having trouble with Wi-Fi,  turns out it's something in my Wi-Fi that I can't put I can't understand yet Port opened and the other stuff in the Wi-Fi suggested list. 

i relented to the newest os, setup with ethernet attached, then add the guest wifi.   Thanks for the follow-up...

cocarrig, 2023-02-14:
Hi @maximillionfrday 

Thanks for the followup, the Pi3 should work well on the latest release, and has a much easier setup than the older 2.25 setup.  Glad you were able to get it working for your setup.  Reach out if you need anything else!