dakboard/Hardware-OS #86: Auto Setting For Screen Resolution

cocarrig, 2023-03-29:
Description:
Creating this case for the purpose of setting the capability to set the display resolution to Auto in DAKboard OS.  This will function for automatic screen configuration in the /boot/config.txt or xrandr for Pi4B systems:
[image]

DaleJP, 2023-04-13:
I do not see any of the ultrawide resolutions on mine even though I is connected to a 21:9 display.

I have the most recent build on a Pi 3b+.  Am I missing something?

cocarrig, 2023-04-13:
@DaleJP,

Looks like it was uncovered today that one of the support articles indicated that the Pi3B+ can use ultrawide, this is unfortunately not the case and the hardware is not capable of ultrawide.  Only the Pi4B Raspberry Pi computer is capable of running ultrawide HD or ultrawide 2k, sorry for the inconvenience this may have caused you.

DaleJP, 2023-04-13:
I see. Well that is incredibly disappointing as I've physically mounted an ultrawide monitor at the front of the school for this purpose 😅
Do you have any other suggestions for hardware that will work with an ultrawide? Pi4s are still really hard to come by here.

cocarrig, 2023-04-14:
@DaleJP,

Sorry, I should have mentioned a few other options for you.  If your device can use ultrawide 2k, then you can use an Orange Pi 3 LTS computer.  It cannot do ultrawide HD as a warning.
This computer is in the [DAKboard CPU v3](https://shop.dakboard.com/collections/dakboard-cpu/products/dakboard-cpu-v3)

You can also download the image if you have an OrangePi 3 LTS just like DAKboard OS from the following support article or purchase the SD card:
https://dakboard.freshdesk.com/support/solutions/articles/35000210494-orange-pi-download-and-install-dakboard-os
https://shop.dakboard.com/products/copy-of-pre-loaded-micro-sd-card-for-orange-pi

cocarrig, 2023-06-08:
Adding a quick note here that display resolution and orientation reporting to dakboard.com will need to be redesigned to query xrandr for current setting, as opposed to reporting the configured variable from the config page(auto) which is not helpful for customer support team.

Dan-Peck, 2023-09-14:
Merged in @ d3e89f6 for upcoming release.

cocarrig, 2023-09-27:
Released in DAKboard-OS 3.52:
https://github.com/dakboard/Hardware-OS/releases/tag/3.52

closed 2023-09-27 by cocarrig