dakboard/Hardware-OS #87: Provide for workaround for internet validation (ping)

cocarrig, 2023-03-29:
Description:
For network validation, the DAKboard OS completes a ping request to validate internet access.  Because some organizations block outbound ping requests there is a use case for the ability for the users with terminal access to change the destination ping address in one location as opposed to the current hardcoded multiple locations.

cocarrig, 2023-05-25:
potential redesign methodology using netcat:
nc -vz dakboard.com 443

success criteria:
'succeeded' is present in response

Dan-Peck, 2023-11-13:
After some initial research, `nc` appears to never exit when there's a failed connection, and while we can specify the length of timeouts, there is no way to specify the number of retries!

However, there is a CLI unix utility called `timeout` that can kill a process after X amount of time.  I've tested this and it works nicely!  The exit status can be used to determine the success of the operation.  If it is `0`, then it was a successful test, anything else, and it is a connection failure.