dakboard/Hardware-OS #99: Sleep schedule cec operation not functional with some devices

cocarrig, 2023-08-30:
**Describe the bug**
Reported via user in support ticket, CEC commands with Visio and potentially other devices does not appear to be operating as expected on RPi and OPi DAKboard OS Systems.  This appears to be generally focused around the CEC standby function from the cec-utils application, more specifically the following command called in argument from the sleep script.

echo standby 0 | cec-client -s -d 1

**To Reproduce**
Steps to reproduce the behavior:
1. Deploy DAKboard OS on RPi or OPi in 3.41 or 103.41 or higher
2. connect to viso TV model number (v555-j01), other models likely to exibit the same behavior
3. Setup a Sleep schedule in the OS configuration page
4. Note that the HDMI will turn off, but the TV will report no signal

**Expected behavior**
The TV should enter standby mode on off transition.

**Screenshots**
N/A

**Device Type (please complete the following information):**
- All DAKboard OS Device Type

**DAKboard OS Version (please complete the following information):**
Newest OS available 3.42 and 103.41

**Additional context**
N/A

cocarrig, 2023-09-11:
Using methods described in the following article, we believe we have been able to resolve this issue by including the following command in the set for the on function.  This sets the TV input to the RPi/OPi and appears to be necessary for multiple transitions:
https://www.linuxuprising.com/2019/07/raspberry-pi-power-on-off-tv-connected.html
`echo 'as' | cec-client -s -d 1`

Dan-Peck, 2023-09-14:
Merged in @ f3aa998 for the upcoming release.

cocarrig, 2023-09-27:
Released in DAKboard-OS 3.52:
https://github.com/dakboard/Hardware-OS/releases/tag/3.52

closed 2023-09-27 by cocarrig