dakboard/Hardware-OS #100: Solve for ability to navigate back to DAKboard OS display URL

cocarrig, 2023-09-18:
Description:
With touchscreen support increasingly supported in the DAKboard OS, the ability to navigate back to a display URL is becoming more important as users click RSS links and buttons that navigate to other locations.

This case is being created to evaluate the options available to restore the DAKboard OS browser back to the main display URL.

drksecretmc, 2024-08-28:
I encountered exactly the same. what is the point of a url or button if you cannot do anything after. Posted a year ago and nothing happened. Disappointing.

cocarrig, 2024-08-28:
@drksecretmc

This is a tough one from a technical standpoint.  Many pages easily allow you to swipe back to the previous page and are not a problem, but identifying those that do not is not possible, and once the browser has left the dakboard.com page we can no longer control the page of course.  Adding the selected page into an iframe is also not always supported, as this is disabled for security purposes on many pages.

At this time we suggest using a URL that you trust, and allows navigation back to the previous page.  Otherwise you can restart the device to restore the page.

drksecretmc, 2024-08-28:
Thank you. But how does it allow to go back? All it opens is a browser
window with no navigation?
......
Sent from my phone—no flashy signature, just a quick response with maybe a
typo or two (thanks, autocorrect). Still just as sincere.

cocarrig, 2024-08-28:
Hi @drksecretmc

I'm assuming your equipment is on touchscreen to interact with this to click the the button of course.  The back gesture via touch would be a long swipe from the left of the screen to the right.

drksecretmc, 2024-08-28:
Hi, no I do not have a touch screen. I have mouse and keyboard. raspberry
pi 5
......
Sent from my phone—no flashy signature, just a quick response with maybe a
typo or two (thanks, autocorrect). Still just as sincere.

cocarrig, 2024-08-29:
Hi @drksecretmc,

That would explain the difficulty, you may be able to press ALT(or command on mac keyboard)+Left Arrow on the keyboard.

cocarrig, 2024-10-30:
Since navigation away from a dakboard hosted site does unfortunately put us at the control of the the destination site's capabilities, the ability to resolve this is limited unfortunately.  At this time we are going to close this case without a fix, and it may be reopened should some enhancements to chromium-browser or other options become more available.

closed 2024-10-30 by cocarrig