dakboard/Hardware-OS #118: Touchscreen unable to click with attached touchscreen IR

cocarrig, 2023-11-02:
**Describe the bug**
Using a USB based IR touchscreen, after first setup and a subsequent reboot the touchscreen does not interact with the browser.  If the USB connection is disconnected and connected again it will work for landscape, but not portrait matrix.  Saving the orientation on the config page appears to resolve issue until the next reboot.

**To Reproduce**
Steps to reproduce the behavior:
1. Deploy 3.53 on Pi 4B
2. Connect IR panel and complete setup in portrait orientation
3. reboot device
4. note that touch no longer interacts with screen for scrolling

**Expected behavior**
Touch function in chromium should persist reboots. 

**Screenshots**
N/A

**Device Type (please complete the following information):**
- Raspberry Pi 4B

**DAKboard OS Version (please complete the following information):**
3.53

**Additional context**
N/A

scottbryant99, 2023-11-02:
I have this issue with 3.53. I downgraded to legacy DAKOS 3.42 to fix the issue until a new DAKOS release addresses this issue.

cocarrig, 2023-11-06:
Current Status:
- This issue appears to have been caused from steps completed to accomodate different touch panels.  Certain touchpanels such as 'ILITEK' touchpanels register as two inputs in xinput, and require the touchscreen device to bee specified in chromium when opened.  Despite selecting the right input device for non 'ILITEK' touchscreens, this action results in the touchscreen not functioning as described in this case.

Next Steps:
The code will need to be updated to ignore and not add any touch device args for chromium, in cases where touch devices are present that do not register as 2 inputs.

cocarrig, 2023-11-10:
Fix completed in latest release:
https://github.com/dakboard/Hardware-OS/releases/tag/3.54

closed 2023-11-10 by cocarrig