dakboard/Hardware-OS #120: Touch features not working on OPi 3LTS running v3.53

jkotecki, 2023-11-05:
I am unable to use the touch setup screen to configure the device. After setting it up via wi-fi, the touch features continue to not work. The cursor responds to touch, but I am unable to select the gear or the next screen icons. These features worked perfectly in v103.41.

cocarrig, 2023-11-06:
Hi @jkotecki,

Can you indicate what kind of touchscreen you are using with this system?

cocarrig, 2023-11-14:
@jkotecki 

We believe the issue you could have been facing was resolved in the following case, and a fix is now available in version 3.54:
https://github.com/dakboard/Hardware-OS/issues/118

We will be closing this case, if you find alternating information or continues issues please reach out.

closed 2023-11-14 by cocarrig