dakboard/Hardware-OS #123: Apply Touchscreen Matrix when touchscreen is connected via USB

cocarrig, 2023-12-01:
Currently, the touchscreen matrix is applied to the touchscreen device on system startup, and when a sleep schedule wakes the attached display.  Reports from clients have indicated that sometimes the usb touchscreen interfaces they use will disconnect and connect again outside of these conditions, and for any orientation besides landscape leave the display with the incorrect matrix.

This case has been created to investigate and build service that will complete the touchscreen routines normally completed on startup, when a touchscreen device is connected.

Dan-Peck, 2023-12-05:
May / may not be helpful, INPUTDEVICES SECTION:

https://www.x.org/releases/current/doc/man/man5/xorg.conf.5.xhtml

Possibly a file like the following added into `xorg.conf.d` directory?

```Section "InputClass"
    Identifier  "calibration"
    MatchProduct    "your product here"
    Option "TransformationMatrix" "1 0 0 0 1 0 0 0 1"
EndSection
```

However, it mentions for hotplugging, this may not be necessary in recent OS's that use HAL or udev backends to handle that.

cocarrig, 2024-06-21:
Due to system impact on smaller Raspberry Pis to restart the chromium process during USB Touchscreen removal/assertion, this approach will be modified to leave the chromium process running now and only reapply the matrix

Dan-Peck, 2024-06-21:
Adding a reminder in here to follow up in order to see if we have tried using that `xorg.conf.d` config file mentioned above.  If we were able to successfully create a config file like that for a given touch device(s), either preconfigured or during Settings-saves, we might not have do do any of the monitoring, restarting, and reconfiguring at all, and let X11 take care of it for us.

jatoric, 2024-07-08:
Released with 4_03 - merged

closed 2024-07-08 by jatoric