dakboard/Hardware-OS #125: Apple Photos not rotating with Dak OS

timrettop, 2023-12-11:
**Describe the bug**
Using same hardware, but with a Raspbian Legacy 32 bit OS setup with chromium kiosk mode, the photos setup with Apple Photo background will rotate as expected, however with the DakOS image, the screen never changes from the initially loaded image. 

**Expected behavior**
Images to rotate every 2 minutes as configured. 

**Device Type (please complete the following information):**
- Raspberry Pi 4B 2GB

**DAKboard OS Version (please complete the following information):**
3.54

**Additional context**
I've attempted adding arm_boost=1 to the boot config.txt, which hasn't changed the issue.

gatorchomp15, 2023-12-15:
> Hi @timrettop,
> Small note for you if you encounter this again, apple photo URLs have had a tendency in recent history to take a very long time to respond.  If you happen to face this, we have other photo integrations such as Google and dropbox that are more reliable.

Just to let you know that I have a dakboard cpu and latest image and screen is resetting itself. Over and over. I’m going to image it over I guess Eric S. FishonSmall Business Owner / Brand Ambassador / Public SpeakerPresident LI Against Domestic Violence | Gator VenturesTogether we can make a ***@***.***://bit.ly/3OSrfee“Envision where your’e headed, honor where you’ve been, and always extend a hand to guide others along the path you have walked. , Your journey today is someone else’s destiney tomorrow” - ESF   https://youtu.be/2y-G2S5kwYM

cocarrig, 2023-12-18:
This can occur if the Micro HDMI on the RPi4B needs to be reseated.  Let me know if that helps.