dakboard/Hardware-OS #138: WiFi connection issue with Open Network

jatoric, 2024-03-13:
**Describe the bug**
When trying to connect DAKboard device on OS 4.01 to networks without security set. Security setting on the DAKboard settings page set to "None". The network fails to connect.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to setting/device network configuration
2. Select Open network SSID
3. Set security type to "None"
4. Save changes.
5. After restart should see "No internet Connection" page.

**Expected behavior**
What should happen - Open network with security setting type "None" should connect without issues. Works with OS version 3.54.

**Device Type (please complete the following information):**
- Any DAKboard devices on OS 4.01

**DAKboard OS Version (please complete the following information):**
 OS version 4.01

jatoric, 2024-03-13:
Removing args being set when "None" type is selected for security type resolves this issue.

cocarrig, 2024-03-28:
Completed and released in 4.02:
https://github.com/dakboard/Hardware-OS/releases/tag/4.02