dakboard/Hardware-OS #143: Screen orientation doesn't apply RPi3

jatoric, 2024-04-01:
**Describe the bug**
Screen orientation doesn't apply on the RPi3 when using unsupported Pi3 resolutions (1920x1200).

**To Reproduce**
Steps to reproduce the behavior:
1. Go to settings page
2. Scroll down to "Screen" 
3. Set resolution to 1920x1200.
4. Change Orientation.
5. Save changes.

**Expected behavior**
Screen Orientation to change after new settings have been saved. 

**Device Type (please complete the following information):**
- Raspberry Pi  3B

**DAKboard OS Version (please complete the following information):**
4.02

jatoric, 2024-04-03:
Descriptions of the bug:
More information from user Miroslav Hrabcak.

When I set manually resolution WUXGA 1920x1200, which is default for monitor, monitor say - The current input timing is not supported by the monitor display.
Monitor is Dell U2410f. When I set Auto, the correct resolution is set. Please see the attached files.

To Reproduce
Steps to reproduce the behavior:

Go to the settings page
Scroll down to "Screen"
Set resolution to 1920x1200.
Save changes.
Monitor stay blank

Device Type:

Raspberry Pi 5

DAKboard OS Version (please complete the following information):
4.02

Screenshots:

[image]
[image]

jatoric, 2024-04-04:
xrandr --current output on the U2410 monitor from a pi5 provided by Miroslav:

```
Screen 0: minimum 320 x 200, current 1200 x 1920, maximum 8192 x 8192
HDMI-1 connected primary 1200x1920+0+0 left (normal left inverted right x axis y axis) 518mm x 324mm
   1920x1200     59.95*+
   1920x1080     60.00    50.00    59.94    24.00    23.98  
   1920x1080i    60.00    50.00    59.94  
   1600x1200     60.00  
   1280x1024     75.02    60.02  
   1152x864      75.00  
   1280x720      60.00    50.00    59.94  
   1024x768      75.03    60.00  
   800x600       75.00    60.32  
   720x576       50.00  
   720x576i      50.00  
   720x480       60.00    59.94  
   720x480i      60.00    59.94  
   640x480       75.00    60.00    59.94  
   720x400       70.08  
   3440x1440_30.00  29.95  
   2560x1080_60.00  59.98  
   2560x1080_30.00  59.98  
   1920x1200_60.00  59.88  
   1680x1050_60.00  59.95  
   1440x900_60.00  59.89  
HDMI-2 disconnected (normal left inverted right x axis y axis)
   3440x1440_30.00  29.95  
   2560x1080_60.00  59.98  
   2560x1080_30.00  59.98  
   1920x1200_60.00  59.88  
   1680x1050_60.00  59.95  
   1440x900_60.00  59.89  
```

jatoric, 2024-04-05:
This issue looks to be isolated to the DELL U2410f monitor - unable to be recreated on other monitors. Closing case.

jatoric, 2024-04-08:
This issue was reported to be for all 16:10 monitors. Testing 16:10 monitor with Pi4/5 - reopening ticket.

nkm8, 2024-11-30:
Chiming in to add that I am also seeing this issue with a 3B+ and a [touch display 2](https://www.raspberrypi.com/documentation/accessories/touch-display-2.html)

jatoric, 2024-12-30:
Found a possible solution for the issue with the change in refresh rate for any RPI before v4 and v5.

jatoric, 2025-01-02:
Fix for issue included with the release of dakboard OS 4.06

jatoric, 2025-01-08:
Fixed with  DAKboard OS 4.06

closed 2025-01-08 by jatoric