dakboard/Hardware-OS #156: CEC not working with scheduled power on

jimpryde, 2024-04-26:
I saw this with version 3.0 of the OS and had reported it before. I rebuilt my Pi 3B+ with the latest version of Dak OS. Set up was fine, but when I do a scheduled power on, the display (an Onn Roku TV) will not switch inputs, despite its internal settings being set to automatically use the last input. If I grab the remote and select the DakBoard input, it works fine. Note: The power up DOES power up the display, but the content is not displayed. 

**To Reproduce**
Steps to reproduce the behavior:
1. Observe LCD display powering up, but remaining on home screen after Pi 3B+ running Dak OS powers on. 
2. Grab remote to TV and switch inputs (which works)
3. Rebooting Pi does not resolve issue
4. Screen does power down at Dak Board scheduled power down time. 

**Expected behavior**
I expect the LCD to power up and automatically begin displaying my Dakboard content. This was addressed in October of 2022 after contacting support. It is the same problem in the new OS. 

**Screenshots**
N/A

**Device Type (please complete the following information):**
Rapberry Pi 3B+
ONN Roku TV

**DAKboard OS Version (please complete the following information):**
 Dak OS 4.0.2

cocarrig, 2024-04-29:
Hi @jimpryde 

The vgencmd service previously used for sleep functions was removed from the Raspbian repo in the Bookworm release on the 4.x releases.  Because of this, a new approach was adopted that was used in the Orange Pi Repo that utilized xrandr functions.  This has been tested pretty extensively and has been in place on the Orange Pi repo, for more than a year as well.

You could revert to 3.54 as a workaround in the meantime:
https://github.com/dakboard/Hardware-OS/releases/tag/3.54

You may run the following to determine if any errors are seen on your install, via SSH while the monitor is in schedule off (I added the first line separately, which may be needed to run via ssh):
```
DISPLAY=:0
echo 'on 0' | cec-client -s -d 1 | grep "opening a connection to the CEC adapter..."
echo 'as' | cec-client -s -d 1
/home/dakboard/startup/init-display.sh
/home/dakboard/startup/init-audio.sh
echo "Display and HDMI have been turned on by user."
DISPLAY=:0 /home/dakboard/startup/init-touch.php
```

cocarrig, 2024-07-08:
Closing this for now.  Can reopen if the requester can provide some additional error information.