dakboard/Hardware-OS #159: Viewsonic TD2760 fails to apply the Touch Matrix when using Sleep Scheduler in Portrait mode

rsalsbery, 2024-04-30:
It was determined that when using the Viewsonic TD2760 touchscreen with the DAKboard OS device if you set it to Portrait mode and use the On/Off Sleep Schedule, it will break the touch matrix when it wakes. This is happening because the TD2760 turns off the power to the Touch Controller when it does not detect an HDMI signal. 

When the DAKboard OS sends the command to restart the HDMI port, the Touch Controller is delayed in starting so the Portrait matrix is applied before it is active. Because this happens it attempts to default to the Landscape orientation,

Jkvette719, 2024-06-25:
Does this happen in landscape? What about using a PC? Assume this is using the Dak CPU?

rsalsbery, 2024-06-25:
@Jkvette719  Inline answers below:

- Does this happen in landscape? No, this only affects portrait mode and Only when using the DAKboard OS On/Off Sleep Scheduler to wake the display. If the Display is in Landscape, when it wakes and starts the USB port it will default to the Landscape matrix and just work.
- What about using a PC? A PC should not be affected as it does not use the DAKboard OS On/Off Sleep Scheduler. However, we have not tested putting the monitor to sleep when connected in portrait mode from a PC. This will depend on how the PC is programmed to handle this. Most likely you would not be able to set the Monitor into portrait mode with a PC unless they have software that allows it available to the PC.
- Assume this is using the Dak CPU? Correct, this only affects devices using the DAKboard OS.

Jkvette719, 2024-06-25:
Wow that was fast.. thanks for the info!

closed 2024-07-08 by jatoric