dakboard/Hardware-OS #164: CPU will not load, keeps looping. On FireTV.

joemayher, 2024-07-15:
Dakboard CPU v4 on FireTV. Worked for a few weeks then for no reason it will not boot up. Dakboard screen shows briefly in attempt to load then shuts off, and attempts a few times in a loop but doesn't load. Tried unplugging both TV and CPU many times but keeps doing the same actions.

cocarrig, 2024-07-15:
Hi @joemayher,

Sorry to hear your're running into that issue, this generally always indicates a problematic connection via HDMI.  Try powering both devices off, then securely reseating the micro HDMI cable end into the CPU and the TV, then power on the TV, then the CPU v4.  Try to avoid hanging the CPU by its cords, as this will normally cause this issue.  If you still have issues, reach out to support at the link below and they can help you out:
https://dakboard.freshdesk.com/support/tickets/new

closed 2024-07-15 by cocarrig

joemayher, 2024-07-16:
It is working so far. Thanks for the tip!