dakboard/Hardware-OS #176: Hang on Loading Screen After Hostname Update

aaronklor, 2024-10-01:
**Describe the bug**
Dakboard OS sits on `Loading...` screen indefinitely after updating hostname using raspi-config

**To Reproduce**
Steps to reproduce the behavior:
1. Have a fully configured Dakboard installation:
    a. Raspberry Pi 4, 4gb
    b. Used Raspberry Pi Imager to install Dakboard OS
    c. Added to Dakboard account using Link Code
    d. Assign screen/loop and confirm is working
2. Enable ssh via "view device settings" in Dakboard UI after "enable ssl"
3. ssh to device using credentials and ip from step 2
4. `sudo raspi-config`
5. Using the cli menus, change hostname from `dakboard` to `this-specific-dakboard`
6. Reboot when prompted
7. Once finished rebooting:
    a. The browser opens and attempts to load Dakboard 
    b. Hangs on a screen containing the Dakboard logo and the text 'Loading...'
    c. Device remains accessible via ssh using either IP or new hostname
    d. GUI elements are **not** frozen

To revert:
1. ssh to device using same credentials
2. `sudo raspi-config`
5. Using the cli menus, change hostname from `this-specific-dakboard` to `dakboard`
6. Reboot when prompted
7. Device boots up and loads Dakboard normally, displaying the expected loop/screen

**Expected behavior**
Changing hostname should not break the dakboard integration.

OR

Changing hostname should give warning message in Dakboard UI

OR 

Changing hostname should prompt user to re-link the device (if it's a registration thing)

**Preferred behavior**
A 'hostname' field in the Advanced settings UI and associated PHP code to perform necessary system- or account-level changes to update the hostname for the device.

**Device Type (please complete the following information):**
- Raspberry Pi 4, 4gb

**DAKboard OS Version (please complete the following information):**
4.0.2

**Additional context**
I have 3 Dakboard devices configured in my home and must distinguish them on the network in order to properly maintain them. Having conflicting hostnames makes this significantly more difficult.

aaronklor, 2024-10-01:
This may be a more specific version of #142

jatoric, 2024-12-04:
Closing this case - working on this issue with #186

closed 2024-12-04 by jatoric