dakboard/Hardware-OS #185: DakBoard Bug

TruFam5280, 2024-12-03:
Upon initial set up an ongoing of the dashboard touchscreen all in one unit purchased within the last month, the screen does not remain on consistently and will glitch on and off constantly

cocarrig, 2024-12-03:
Hi @TruFam5280,

Sorry to hear you're having issues with your hardware.  If this was a DAKboard sold Wall Display please create a ticket with the support team and we'd be glad to work to resolve this issue for you:
https://dakboard.freshdesk.com/support/tickets/new

If this was a reseller item alternatively, we would suggest contacting them through their official channels and they should be able to help you alternatively.

For now we'll close this bugs case unless there are some additional software bug details we can work to resolve.  If that is the case please feel free to reopen.

closed 2024-12-03 by cocarrig

gatorchomp15, 2024-12-06:
CoryYou definitely have much on your plate these days as I see the emails flying back and forth. I hope you remember me I’m on your beta team, have the mini boxes, five raspberry and all. So over the past month I’ve transitioned from my touch screen I had for something more versatile that I’ve put dakboard on and people are asking so many questions. You know I love your company and as an ambassador I want to stay up to the latest and greatest updates. So I have a Coyla screen which you’ve probably heard of it’s basically a touch display which could be either in a 15 inch or 22 inch or 32 inch either on the wall or on wheels and it’s basically a the lack of a better word one big android device that you can pull in your own calendars stream watch TV but it’s really supposed to be made for integration with your own to-do list your own calendars and so forth. One of the big messes I swear that it had was really putting in your creation of your own type of calendar the way you wanna see it and that’s when I put my dashboard on there as a static screen and people just loved it because putting the Christmas countdown on like was already shown and so I’m hoping that you can get a surge of some new members. Is there a certain link you want me to give people that goes to purchase your products mainly purchasing your membership. Also I am going to do a video as well. You Dr. Eric S FishonDr. Eric Fishon | MHA/ Owner / Chief XTERMIGATORa: ABCO PEST CONTROL  | Long Island, NY | 4th Generation Veteran & Family ***@***.*** | w: www.abcopest.com m: 646-235-5991 | p: 631-979-PEST    PresidentA Disability is not an Inability Together we can make a Difference“A disability is not an inability; it is an opportunity to adapt, grow, and demonstrate strength in ways that others may never experience.” ~ Eric Fishon '

https://dakboard.freshdesk.com/support/tickets/new

cocarrig, 2024-12-06:
Hi Eric,

Hope you're doing very well.  For now you can simply send them to the normal dakboard.com domain, but keep an eye on the blog.dakboard.com site as we will announce an affiliate program hopefully very soon.  Hope you're having a great holiday season!

gatorchomp15, 2024-12-09:
On the community that I’m on for Cozyla i’m getting a lot of questions from users asking where the deck board app went from the Google play store. I didn’t want to give them wrong information if you wouldn’t mind.Eric S. Fishon, DHA(c)Entrepreneur / Brand Ambassador  / Public SpeakerPresident LI Against Domestic Violence | Gator VenturesTogether we can make a ***@***.***://bit.ly/3OSrfee“Envision where your headed, honor where you’ve been, and always extend a hand to guide others along the path you have walked. , Your journey today is someone else’s destiny tomorrow” - ESF   https://youtu.be/2y-G2S5kwYM

cocarrig, 2024-12-10:
Hi @gatorchomp15,

Due to some android requirements, we were forced to take down the App or redesign it.  Our focus lately has been on Raspberry Pi, but we do have a future goal of providing another Android App.  At the moment we dont have a timeline for providing this yet.

gatorchomp15, 2024-12-10:
This is why I appreciate this channel it helps to provide the information we need. Eric S. Fishon, DHA(c)Entrepreneur / Brand Ambassador  / Public SpeakerPresident LI Against Domestic Violence | Gator VenturesTogether we can make a ***@***.***://bit.ly/3OSrfee“Envision where your headed, honor where you’ve been, and always extend a hand to guide others along the path you have walked. , Your journey today is someone else’s destiny tomorrow” - ESF   https://youtu.be/2y-G2S5kwYM