dakboard/Hardware-OS #192: RPI ZERO 2 Modification script

Jesstr8803, 2024-12-16:
Hello, 
Here is a little script I worte with some modifications.  I am not a programmer, Chat GPT helped me though most fo this. It does a few things.

I've been using Dakboard with a USB touch screen, just a cheap one from Amazon, it does not support CEC, and when the HDMI signal is disconnected it will show like a test screen that keeps the screen on, flipping though colors. I modified the rpi-hdmi.sh script which gets called when scheduled from the cron jobs. This script is what blanks the screen. I instead have the script just change the brightness of the connected display. This seems to be much better. It might be worth adding an option checkbox in the future for displays that dont have CEC.

The script will then change a few of the config.txt file settings. Bigest change being the GPU_MEM usage. I found that a little higher setting made the display much more fluid.

The last thing the script does is change the touchscreen logic. I had a hell of a time getting the touch screen settings to work with dakboard. This script activly monitors the orientation of the display and updates the touchscreen matrix. There is also a service file that is created to run the script. 

Again, I am not a programer, I don't know if any of this was a good idea. I don't know if the rpi-hdmi.sh file will be overwrote by updates. But this has helped me with my setup!

If you want to use this file:
1. Create a blank .sh file, copy code contents to it.
2. Copy .sh file into boot partiton.
3. login with ssh, navigate to boot partition, make file excutable.
4. Run file. 

I will not be able to support anyone who tries this code, use at own risk. Developers are more than free to use any part of this.

Thank you,
Jesse

`#!/bin/bash

# Paths to important files
RPI_HDMI_SCRIPT="/home/dakboard/rpi-hdmi.sh"
CONFIG_FILE="/boot/config.txt"
MONITOR_AND_ADJUST_SCRIPT="/usr/local/bin/monitor_and_adjust.sh"
MONITOR_AND_ADJUST_SERVICE="/etc/systemd/system/monitor-and-adjust.service"

# Backup original files
echo "Backing up existing files..."
sudo cp $RPI_HDMI_SCRIPT ${RPI_HDMI_SCRIPT}.backup.$(date +%F-%T)
sudo cp $CONFIG_FILE ${CONFIG_FILE}.backup.$(date +%F-%T)

# Apply the new rpi-hdmi.sh script
echo "Updating rpi-hdmi.sh script..."
cat << 'EOF' | sudo tee $RPI_HDMI_SCRIPT > /dev/null
#!/bin/sh

# Script to control display brightness for HDMI

set_brightness() {
    local brightness=$1
    DISPLAY=:0 xrandr --output HDMI-1 --brightness "$brightness" > /dev/null 2>&1
}

case $1 in
    on)
        set_brightness 1
        echo "Display brightness set to maximum."
        ;;
    off)
        set_brightness 0
        echo "Display brightness dimmed to minimum."
        ;;
    status)
        current_status=$(xrandr --verbose | grep -A 5 "HDMI-1" | grep "Brightness" | awk '{print $2}')
        echo "Current brightness level: $current_status"
        ;;
    *)
        echo "Usage: $0 on|off|status" >&2
        exit 2
        ;;
esac

exit 0
EOF

# Make the rpi-hdmi.sh executable
echo "Setting permissions for rpi-hdmi.sh..."
sudo chmod +x $RPI_HDMI_SCRIPT

# Apply the new config.txt file
echo "Updating config.txt..."
cat << 'EOF' | sudo tee $CONFIG_FILE > /dev/null
disable_overscan=1
framebuffer_depth=24
gpu_mem=128
hdmi_force_hotplug=1
hdmi_group=2
hdmi_mode=69
dtparam=audio=on
avoid_warnings=1
dtoverlay=gpio-shutdown
disable_splash=1
EOF

# Create the monitor and adjust script
echo "Creating monitor and adjust script..."
cat << 'EOF' | sudo tee $MONITOR_AND_ADJUST_SCRIPT > /dev/null
#!/bin/bash

# Variables
DISPLAY_READY=0
LAST_ROTATION=""
LOG_FILE="/tmp/monitor_and_adjust.log"
DEBOUNCE_DELAY=2  # Time in seconds to wait before processing another orientation change

# Initialize log file
echo "Starting monitor_and_adjust.sh at $(date)" > $LOG_FILE

# Check if DISPLAY is set
if [ -z "$DISPLAY" ]; then
    export DISPLAY=:0
    echo "DISPLAY variable set to :0" >> $LOG_FILE
fi

# Wait for X server to be ready
echo "Checking X server readiness..." >> $LOG_FILE
while ! xrandr --query >/dev/null 2>&1; do
    echo "X server not ready. Waiting..." >> $LOG_FILE
    sleep 5
done
echo "X server is ready." >> $LOG_FILE

# Wait for touchscreen input devices to be ready
echo "Checking touchscreen devices..." >> $LOG_FILE
while ! xinput list | grep -i "touchscreen" >/dev/null; do
    echo "Touchscreen devices not ready. Waiting..." >> $LOG_FILE
    sleep 5
done
echo "Touchscreen devices are ready." >> $LOG_FILE

# Monitor for orientation changes
echo "Starting orientation monitoring..." >> $LOG_FILE
while true; do
    # Get the current rotation of the display
    CURRENT_ROTATION=$(xrandr --query | grep "HDMI-1 connected" | awk '{print $5}' | tr -d '()')

    # If the rotation changes and is stable, adjust the touchscreen input
    if [ "$CURRENT_ROTATION" != "$LAST_ROTATION" ]; then
        echo "Detected potential orientation change: $CURRENT_ROTATION" >> $LOG_FILE
        sleep $DEBOUNCE_DELAY  # Wait to ensure the change is stable
        NEW_ROTATION=$(xrandr --query | grep "HDMI-1 connected" | awk '{print $5}' | tr -d '()')

        if [ "$NEW_ROTATION" = "$CURRENT_ROTATION" ]; then
            echo "Confirmed stable orientation: $NEW_ROTATION" >> $LOG_FILE
            LAST_ROTATION="$NEW_ROTATION"

            # Adjust touchscreen input for the new orientation
            TOUCH_IDS=$(xinput list | grep -i "touchscreen" | grep -oP 'id=\K\d+')

            for ID in $TOUCH_IDS; do
                echo "Adjusting touchscreen with ID $ID for $NEW_ROTATION orientation" >> $LOG_FILE
                case $NEW_ROTATION in
                    normal)
                        xinput set-prop $ID "Coordinate Transformation Matrix" 1 0 0 0 1 0 0 0 1 || \
                        echo "Error adjusting ID $ID for normal orientation" >> $LOG_FILE
                        ;;
                    inverted)
                        xinput set-prop $ID "Coordinate Transformation Matrix" -1 0 1 0 -1 1 0 0 1 || \
                        echo "Error adjusting ID $ID for inverted orientation" >> $LOG_FILE
                        ;;
                    left)
                        xinput set-prop $ID "Coordinate Transformation Matrix" 0 -1 1 1 0 0 0 0 1 || \
                        echo "Error adjusting ID $ID for left orientation" >> $LOG_FILE
                        ;;
                    right)
                        xinput set-prop $ID "Coordinate Transformation Matrix" 0 1 0 -1 0 1 0 0 1 || \
                        echo "Error adjusting ID $ID for right orientation" >> $LOG_FILE
                        ;;
                    *)
                        echo "Unknown rotation: $NEW_ROTATION. Skipping adjustment for ID $ID." >> $LOG_FILE
                        ;;
                esac
            done
        else
            echo "Orientation change not stable. Ignoring transient state." >> $LOG_FILE
        fi
    fi

    # Wait for a short interval before checking again
    sleep 2
done
EOF

# Make the monitor and adjust script executable
echo "Setting permissions for monitor and adjust script..."
sudo chmod +x $MONITOR_AND_ADJUST_SCRIPT

# Create the systemd service for monitoring and adjusting
echo "Creating monitor and adjust systemd service..."
cat << 'EOF' | sudo tee $MONITOR_AND_ADJUST_SERVICE > /dev/null
[Unit]
Description=Monitor Screen Orientation and Adjust Touchscreen
After=graphical.target

[Service]
ExecStartPre=/bin/sleep 45
ExecStart=/usr/local/bin/monitor_and_adjust.sh
Restart=always
User=dakboard
Environment=DISPLAY=:0
Environment=XAUTHORITY=/home/dakboard/.Xauthority

[Install]
WantedBy=default.target
EOF

# Enable and start the systemd service
echo "Enabling and starting the monitor and adjust service..."
sudo systemctl enable monitor-and-adjust.service
sudo systemctl start monitor-and-adjust.service

# Reboot to apply changes
echo "Rebooting to apply changes..."
sudo reboot
`

Jesstr8803, 2024-12-18:
Sorry for creating this in issues, please close and/or delete.

cocarrig, 2024-12-18:
No problem, I see you added this into discussions, so I will close this for now.

closed 2024-12-18 by cocarrig