dakboard/Hardware-OS #195: CEC not properly working with scheduled power on after update to 4.05

stthiele, 2024-12-26:
**Describe the bug**
I initiated an update to version 4.05 on Dec 24, and the whole setup worked as expected without any errors. After adding the WiFi and activating SSH, the Raspberry Pi restarted successfully and the display showed the expected screen loop. I then added two On/Off schedules.

I have been able to confirm the “Off” schedule works, but the “On” has failed to activate for two straight days. The current setup is exactly the same as the pre-update (unknown DAK version), but pre-update the display activated as expected. The display can be manually turned on and input manually selected with successful screen/loop displayed. The display is a multi-use display, in that DAKBoard screens/loops are displayed as well as use of other display inputs. Therefore the display’s stored ‘last input’ could vary from day to day. Prior to the update to 4.05 this was not a problem and the DAKBoard activated the display and updated the input as expected.

**To Reproduce**
Steps to reproduce the behavior:
1. Observe target display prior to scheduled “On” as off. After passing of expected time +2min the display did not turn on.
2. Using display remote, display is turned on and the correct input is selected. The expected screen/loop is delivered.
3. The DAKRPi was rebooted via SSH and new Scheduled On/Off added for top of the next hour. New “On” time passes without turning on.
4. The DAKRPi had the power cord removed/returned with new Scheduled On/Off added for top of the next hour. New “On” time passes without turning on.
5. The HDMI output was changed to the other port and the DAKRPi was rebooted via SSH. A new Scheduled On/Off added for top of the next hour. New “On” time passes without turning on.
6. The display correctly turns off at scheduled “Off”.

**Expected behavior**
I expect the display to power up, select the correct input, and automatically begin displaying my Dakboard content

**Device Type (please complete the following information):**
Display: Vizio M55Q6-J01
DAKBoard: DAKboard CPU v4

**DAKboard OS Version (please complete the following information):**
4.05

**Additional context**
I found thread [https://github.com/dakboard/Hardware-OS/issues/156](https://github.com/dakboard/Hardware-OS/issues/156) and attempted a forced activation using the support information. The display is confirmed to be support 2160p, and the display is set to support this resolution (see screenshot).
[image]

The display does activate and shows the expected screen/loop, but errors are reported. Here is result from the ssh:

> dakboard@dakboard:~ $ DISPLAY=:0
> echo 'on 0' | cec-client -s -d 1 | grep "opening a connection to the CEC adapter..."
> echo 'as' | cec-client -s -d 1
> /home/dakboard/startup/init-display.sh
> /home/dakboard/startup/init-audio.sh
> echo "Display and HDMI have been turned on by user."
> DISPLAY=:0 /home/dakboard/startup/init-touch.php
> opening a connection to the CEC adapter...
> opening a connection to the CEC adapter...
> access control disabled, clients can connect from any host
> access control disabled, clients can connect from any host
> X Error of failed request:  BadName (named color or font does not exist)
>   Major opcode of failed request:  140 (RANDR)
>   Minor opcode of failed request:  16 (RRCreateMode)
>   Serial number of failed request:  37
>   Current serial number in output stream:  37
> X Error of failed request:  BadName (named color or font does not exist)
>   Major opcode of failed request:  140 (RANDR)
>   Minor opcode of failed request:  16 (RRCreateMode)
>   Serial number of failed request:  37
>   Current serial number in output stream:  37
> X Error of failed request:  BadName (named color or font does not exist)
>   Major opcode of failed request:  140 (RANDR)
>   Minor opcode of failed request:  16 (RRCreateMode)
>   Serial number of failed request:  37
>   Current serial number in output stream:  37
> X Error of failed request:  BadName (named color or font does not exist)
>   Major opcode of failed request:  140 (RANDR)
>   Minor opcode of failed request:  16 (RRCreateMode)
>   Serial number of failed request:  37
>   Current serial number in output stream:  37
> X Error of failed request:  BadName (named color or font does not exist)
>   Major opcode of failed request:  140 (RANDR)
>   Minor opcode of failed request:  16 (RRCreateMode)
>   Serial number of failed request:  37
>   Current serial number in output stream:  37
> X Error of failed request:  BadName (named color or font does not exist)
>   Major opcode of failed request:  140 (RANDR)
>   Minor opcode of failed request:  16 (RRCreateMode)
>   Serial number of failed request:  37
>   Current serial number in output stream:  37
> xrandr: cannot find mode 3840x2160
> 536870916
> Display and HDMI have been turned on by user.

Please advise if other troubleshooting or information is needed.

jatoric, 2025-01-10:
@stthiele Would you be able to test this with DAKboard os 3.54.  You can download 3.54 [HERE](https://drive.google.com/file/d/1aiCWSezt7uVE-GDYE2bDpMaSxsuYPwTr/view?usp=drive_link)

We have a Vizio TV of a similar model and have been experiencing significant issues since it has undergone recent Vizio updates.

Let me know if version 3.54 resolves this issue for you.

stthiele, 2025-01-12:
I have downloaded the package, and will get it setup to run. Will report back.

In an update with the current DAKBoard version, the Vizio has been properly activating and changing to the correct input but inconsistently - maybe 2 or 3 on 5 scheduled prompts. There have been some instances where it activates the TV but the TV changes to the Home Input (settings indicate Last Used).

jatoric, 2025-01-28:
@stthiele unfortunately, we are finding the same issues with a similar Vizio model in our office. This appears to be due to frequent updates and advertisements that are regularly shown on the display. At the moment, we do not have a clear development path for this since it appears to be due to the TV configuration. We will be resolving this case, but if a path for resolution is made available to us, this case can be reopened.