dakboard/Hardware-OS #198: Hide or Show Completed does not propagate to the device.

pmcgahan75, 2025-01-27:
I use the DAKboard ToDo feature to enter tasks.  When working on my PC I can show or hide completed tasks, But when I refresh the screen it does not propagate to my connected device.

Steps to reproduce the behavior:
1.  Go to the connected calendar and add a new calendar event to verify the screen refreshes on the connected device
2. Log into **dakboard.com** on the PC
3. In the Menu on the left,  click on 'ToDo'
4. Open up a ToDo list that has some completed tasks
5. Click on the 3 dots to the right of the task list title
6. Toggle the option to 'Hide Completed'
7. Verify that the completed tasks are now hidden on the ToDo list
8. Select 'Screen'  on the dakboard menu and then Refresh or Reload 
9. Now check the monitor connected to the Dakboard and 
      (a) verify that the new calendar event is visible, verifying that the refresh happened.
      (b) look at the ToDo list and note that the the Completed tasks are still visible.

**Expected behavior**
The task list on the connected device will no longer show the completed tasks 

**Screenshot**
[image]

**Device Type (please complete the following information):**
- DAKboard CPU:  DAKboard CPU v4 ×1, Raspberry Pi 4B
- DAKboard Wall Display: not using

**DAKboard OS Version (please complete the following information):**
 If known, please include the DAKboard OS version.

rsalsbery, 2025-01-31:
Sorry for the confusion on this. The Hide Completed you are referring to only affects the Completed events on your To-Do tab and does not propagate to the Screen or Display. To Hide completed on your Screen or Display you must turn off the Show Completed option on your To-Do block editor. See below:

[image]

closed 2025-01-31 by rsalsbery

pmcgahan75, 2025-01-31:
Thank you

 

image.png (view on web) <https://github.com/user-attachments/assets/eff6c99c-dcbe-468d-8390-985219d133d7>