dakboard/Hardware-OS #199: ToDo List Items Not Showing Up

aramritu, 2025-01-27:
**Describe the bug**
Some ToDo items are not showing up when sorting list through due date

**To Reproduce**
Steps to reproduce the behavior:
After some testing found that the creation date of a list item stops from that item from showing up, after recreating items with matching to do date, item shows up.

I am unable to find the exact timing of creation date that allowed an item to show up.

**Expected behavior**
Creation date of a list item stops that item from showing up, after recreating items with matching to do date, item shows up.

**Screenshots**
ToDo List in Microsoft has 18 items, however in Dakboard it showed only 6 items, had to recreate to get these items, I didnt get a screenshot can update this bug when another item does show up that isnt in dakboard

**Device Type (please complete the following information):**
Raspberry Pi 3B+

rsalsbery, 2025-01-31:
Please open a support ticket by clicking on the Chat Icon and typing **Create Support Ticket** to the AI Chatbot. The Chatbot will walk you through the creation steps. Please make sure to include your Account Username or the email you used when you created your account so that we can review the configuration of your To-Do block. This way we can determine if the issue you are describing is a bug of a configuration error.

cocarrig, 2025-01-31:
Resolving for now, but this may be reopened if needed.

closed 2025-01-31 by cocarrig

aramritu, 2025-02-01:
> Please open a support ticket by clicking on the Chat Icon and typing **Create Support Ticket** to the AI Chatbot. The Chatbot will walk you through the creation steps. Please make sure to include [...]

What catergory do I use here? nothing coming up for technical

aramritu, 2025-02-01:
> What catergory do I use here? nothing coming up for technical

Got it now, thought you were speaking about githubs chat and not within dakboard