dakboard/Hardware-OS #205: Calendar block glitching

tarahl1, 2025-02-12:
Calendar block flashing at the top of the screen.

My calendar has been working for a few months now. Today randomly the top 3rd of the calendar block is glitching.
I’ve tried re-adjusting the fit and confirmed aspect ratio, re-adding the display and restarting the system.

 The touch screen is used only to scroll the calendar ahead and has always been slow. 

Is it possibly an update could have affected the Pi?

Any help would be appreciated!

- RaspberryPi 3B+
- 22” touch screen display

cocarrig, 2025-02-13:
Placing in project and assigning to attempt to recreate issue

cocarrig, 2025-02-14:
Hi @tarahl1,

Depending on your DAKboard-OS release installed on your RPi3B+, there could be an issue with the browser settings that was resolved recently.  If you are not yet on the latest release, we would recommend upgrading using the following steps:
https://dakboard.freshdesk.com/support/solutions/articles/35000125880-raspberry-pi-download-and-install-the-dakboard-os

We will resolve this for now, but if this does not resolve your issue please reopen this case.

closed 2025-02-14 by cocarrig