dakboard/Hardware-OS #209: Flashing stripes during boot up process on RPi4B on 4K displays

Dimasua, 2025-02-19:
Hello Dakboard Support Team,

we have addressed this issue a while ago, but so far no solution is provided.

Started from v3.52 we are facing the flashing stripes issues on all RPi4B connected with 4K displays. All FW versions after v3.52 are having this issue, up to 4.07 which we just tried out today. All FW versions before 3.53 do not demonstrate that behavior.

Please see attached screen shots with the stripes and a model of the display we use.

As advised downgrading the quality of HDMI cables didn't help.

Due to this issue all our customers are tied up with v3.52 and can not benefit from the latest releases.

Kind regards,
Dima.

[image]
[image]

cocarrig, 2025-02-19:
Hi @Dimasua,

Since this appears to be isolated to this display, I think we'll need to do direct testing on one.  If you want to contact us via support ticket and share with us some options to get that hardware we can consider that.

If you install a normal latest Raspbian on the Pi as well I suspect you are also seeing the issue?

cocarrig, 2025-03-07:
Per support ticket discussion, this appears to happen on base raspbian release as well and is isolated to this display.  Sine our options are pretty limited to recreate this, and we will need to continue to use raspbian we will be closing this case for now, but would be glad to discuss this more via the support ticket we have with you with additional detail.

closed 2025-03-07 by cocarrig