dakboard/Hardware-OS #215: Dakboard Upside Down / Device is offline

DanielTB80, 2025-04-28:
**Describe the bug**

We have noted that the Dakboard is randomly having the following issues occur:
1. The Dakboard screen presentation flips upside down
2. the Dakboard screen displays an error message: “Dakboard Device is offline - Please check network connectivity!” (Displayed upside down in the left bottom corner)

Upon reboot, the dashboard reboot, upside down and displays “No Internet connection”.

When the device configuration is accessed:
1. The WiFi passcode no longer matches the original saved passcode
2. the display shows that it is flipped horizontally.
3. the On/Off schedule has changed the custom daily schedule back to all days.

Once the WiFi passcode is resaved to the original, correct value, the screen is flipped back and the custom days schedule is reinput, everything is ok for a few days until it does it again.

This started after our we upgraded our wireless access points from a Linksys WiFi 5 device using a single SSID with WPA2 to a Firewalla WiFi 7 device using multiple SSIDs and this device is connected to SSID using only WPA2 (the SSID is not using mixed WPA2/WPA3).

The WiFi device is in the same location as before.

We first encountered this error on OS 4.01, upgraded to OS 4.07 and encountered the same behavior.

**To Reproduce**
Steps to reproduce the behavior:
1. Unable to force error

**Expected behavior**
A clear and concise description of what you expected to happen.
The device should not lose signal, flip screen, change WiFi code and remove on/off custom daily schedule.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device Type (please complete the following information):
- Raspberry Pi 4

**DAKboard OS Version (please complete the following information):**

[image]
[image]
[image]

DAKboard OS 4.01, 4.07

**Additional context**
Add any other context about the problem here.

rsalsbery, 2025-05-02:
Sorry for the confusion on this. To access a DAKboard OS device, a user must either be on the same network as the device or have physical access to the device. However, some vulnerability scans, like what can be done with a FireWalla device, may cause orientation changes and even overwrite passwords on your Display.

To prevent unwanted changes, you can enable SSL on your DAKboard OS device. See our article [DAKboard OS SSL/SSH Configuration and Use](https://dakboard.freshdesk.com/support/solutions/articles/35000222207-dakboard-os-ssl-ssh-configuration-and-use) to see how to enable this. Once this is turned on, to access or make changes to the DAKboard OS device, you will need to use the password you set and the username dakboard when logging into this device to make changes.

DanielTB80, 2025-05-17:
After making the recommended change, we are not seeing this issue reoccur.

Recommend keeping this case open a bit longer to confirm resolution.

closed 2025-05-29 by jatoric